skiselev/isa-super-vga #5: TVGA9000i ISA vs MCA pinout

kladogen, 2024-12-23:
Hi, I have built your design - thanks allot for it!! I have purchased 3 TVGA9000i chips but the board is not working, I have checked everything multiple times. So I just found that the TVGA9000i comes in two pinouts ISA and MCA, how can I check that the chips I have are ISA?

skiselev, 2024-12-23:
Any TVGA9000i chip can work either in ISA or MCA mode. The mode selected by this ISA SVGA board is ISA mode.
Sorry to hear your board is not working.
Do you get any signs of life at all? E.g. BIOS beeps? (If I recall correctly, there is a series of three or so short beeps if VGA memory test fails)
One common issue is soldering of the TVGA9000i chip, I'd recommend making sure that there are no solder bridges between pins. Also sometimes, pins are not thoroughly soldered. Both issues can be checked using a multimeter with thin, needle probes.

Hope this helps...

skiselev, 2025-01-10:
No response in a few weeks, closing the issue. Feel free to reopen or create another issue.

closed 2025-01-10 by skiselev