skiselev/isa-super-vga #7: Crystal casing comes into contact with pads

kladogen, 2025-01-01:
The reason my board did not work was that the crystal case is conductive and came into contact with the pads under the crystal when the crystal is pushed to come in contact with the PCB (us it is supposed to). This causes a shortcircuit between the crystal pins and off course the circuit does not work. The solution is to leave to some space between the crystal and the board or to put some insulating tape under the crystal. In next version of the PCB the crystal pads could become smaller.

wrljet, 2025-01-01:
I don't have that issue with mine.  But, Digi-key, etc. sell little mylar insulators designed to go underneath standard crystal footprints.  I used to use these on my cheap PCBs years ago before I had solder mask.

kladogen, 2025-01-01:
Maybe a comment could be added to the BOM to save others from the same problem. It took me a week to find it after checking all connections on the PCB. The crystal was the last component I checked. At least it worked for 2025!

skiselev, 2025-01-10:
@wrljet do you have a part number, perhaps I can add that do the BoM?

wrljet, 2025-01-11:
Bivar (the manufacturer)  CI-192-028-SR

DigiKey  492-1039-ND
Mouser 749-CI-192-028-SR 

But for what it's worth, I don't see a problem with the PCB causing a need for these insulators.
The expected crystal package won't short anything.

Bill

skiselev, 2025-01-27:
Thanks @wrljet!
I added the insulator to the BoM: https://github.com/skiselev/isa-super-vga/commit/ec1888477d16e3e7cb4c25714abd4655e719a455

closed 2025-01-27 by skiselev