skiselev/isa-super-vga #9: Bracket dimensions seem to be incorrect

skiselev, 2025-01-27:
@giobbino Can you please check the bracket dimensions?
They should be the same as or very close to [Keystone Electronics 
9200-1](https://www.mouser.com/datasheet/2/215/200-1-741899.pdf).
Looks like the position of the connector is not correct.
The center of the connector should be 49.83 mm from the top (50 mm is close enough). In your design it seems to be 55 mm or so.

giobbino, 2025-01-29:
I'm not sure why it's wrong for you, since it worked for me. However not a big deal, I can change the project and add a new file to the page. 

Just to be sure: should I move the VGA connector hole 5 mm upward?

kladogen, 2025-01-29:
If the VGA connector is move 5mm up then the bracket corner should be at the correct height for a standard PC.

giobbino, 2025-01-29:
https://www.thingiverse.com/thing:6473572/files
added the new STL connector 
cheers